RoseauTechnologies/Roseau_Load_Flow #16: Phases management

benoit9126, 2022-12-15:
In order to remove the `n` argument to constructor of classes, we will use phases as strings. Here is the definition of these new arguments:
- [ ] `Bus`: 
  - [ ] the allowed phases are "abc", "abcn", "ab", "bc", "ca", "an", "bn", "cn", "abn", "acn", "bcn". 
  - [ ] This argument of the constructor is mandatory.
- [ ] `VoltageSource`: 
  - [ ] the allowed phases are the same as for the buses. 
  - [ ] Nevertheless, it is not required. By default, the phases of the bus are taken
  - [ ] The phases must be present in the bus it is connected to.
- [ ] `AbstractLoad`: idem as voltage source. NB: When the phase contains a "n", the load will be a star-load, otherwise a delta-load.
- [ ] `Line`:
  - [ ] Only a single `phases` argument to the constructor. The allowed values are "abc", "abcn", "ab", "bc", "ca", "an", "bn", "cn", "abn", "acn", "bcn", "a", "b", "c", "n".
  - [ ] Nevertheless, it is not required. The intersection of the phases of the two buses at the extremities of the line are taken by default.
  - [ ]  The phases of the line must be present in the upstream and in the downstream buses.
  - [ ] The size of the matrices of the line characteristics must be equal to the number of phases of the line.
- [ ] `Switch` ⇾ Same as `Line`
- [ ] `Transformers`
  - [ ] The arguments `phases1` and `phases2` must be added in the constructor of the class.
  - [ ] For the moment, only the phases "abc" and "abcn" are accepted
  - [ ] It is not mandatory to define them. The windings of the transformer characteristics define the default: "D" ⇾ "abc", "Y" ⇾  "abcn"
  - [ ] The `phases1` must be present in the upstream bus.
  - [ ] The `phases2` must be present in the downstream bus.
  - [ ] The winding of the transformer characteristics must be compliant with the selected `phases1` and `phases2`.
- [ ] `Ground`
  - [ ] The constructor only have an `id` argument
  - [ ] The `Ground` class has a method `connect which takes two arguments: a) the bus to connect to; b) the phase to connect to
  - [ ] The `phase` argument of the `connect` method only accepts "a", "b", "c" or "n" as argument.
  - [ ] Its default value is "n".
  - [ ] The chosen phase must be available in the bus.
- [ ] `PotentialRef`
  - [ ] Its constructor takes an element which is a bus or a ground instance. 
  - [ ] Its constructor takes a `phase` argument which can be `a`, `b`, `c` or `n` or `None`.
  - [ ] The default value of the `phase` argument is `None`
  - [ ] If the provided element is a ground, its potential is fixed at 0V. 
  - [ ] If the provided phase is `None` and if the bus' phases contains "n", we fix the potential of the neutral of the bus to zero. Otherwise, we define the sum of potentials of the bus to 0V
  - [ ] If the provided phase is not `None`, we check that this value is present in the bus' phase and we fix it to 0V
  - [ ] In the JSON export, the value provided in the constructor must be serialized (`None` -> `null`)

Saelyos, 2022-12-16:
> `VoltageSource`
> - [ ] The phases must be present in the bus it is connected to.

I've realized that it is too restrictive, while it is possible to connect a delta voltage source on a star bus, it should also be possible to connect a star voltage source on a delta bus. So we could change the condition to : 

> The phases except "n" must be present in the bus it is connected to.

Same for the loads.

benoit9126, 2022-12-16:
If you prefer, it is not a problem