RoseauTechnologies/Roseau_Load_Flow #159: BUG: res_transformers.power1 contains NaN for neutral

YassineAbdelouadoud, 2024-01-09:
**Describe the bug**
The real value of res_transformers.power1 is NaN for the neutral branch.

**To Reproduce**
Minimal Working Example to understand the problem

```python
en = ElectricalNetwork.from_catalogue(name="LVFeeder38211", load_point_name="Winter")
for name, branch in en.branches.items():
    if name == 'Transformer':
        branch.parameters.max_power = 160. * 1e3 * 1.1

en.solve_load_flow(auth=auth)
print(en.res_transformers.power1.isna().sum())
```

**Expected behavior**
The real value of res_transformers.power1 is not NaN for the neutral branch.

**Version details**
roseau-load-flow 0.6.0

```
System Information
------------------
python               3.10.8.final.0
os                   linux
os_name              posix
machine              x86_64
Installed Dependencies
----------------------
pandas               2.1.0
numpy                1.23.5
geopandas            0.13.2
shapely              2.0.1
regex                2022.10.31
pint                 0.22
requests             2.31.0

```

benoit9126, 2024-01-09:
@YassineAbdelouadoud @alihamdan will take a look at this bug and at #158. Thanks for the report. :+1:

alihamdan, 2024-01-10:
Hello Yassine, thanks for the report.
This is working as expected here. The transformer is `Dyn`, it has the phases "abc" on the primary side and "abcn" on the secondary side. Since the neutral doesn't exist on the primary, it has the values in the results set to `nan`. If you don't care that a value does not exist, I suggest filling it with `0` (like in `en.res_transformers.fillna(0)`).
Note that this could only happen with transformers as the phases on the primary and secondary are not guaranteed to be the same, while lines and switches would always have the same phases on both sides.

closed 2024-01-24 by benoit9126