RoseauTechnologies/Roseau_Load_Flow #169: ENH: make the limit on bus number a parameter

YassineAbdelouadoud, 2024-01-22:
**Describe the feature you want. What problem does it solve?**
Currently, the limit of bus number for a simulation is set to 1000 and, to the best of my knowledge, cannot be configured by the user. When modelling MV and LV networks together (example : one MV feeder and the connected LV feeders), this limit is frequently surpassed. I propose to make this limit a parameter that the user can set, at least for the future local version if not the SaaS one. 

**Describe alternatives you've considered**
I am currently only working with networks with fewer than 1000 buses for testing purposes.

**Additional context**
For a random sample of 100 MV feeders, there are around 2/3 of networks with more than 1000 buses. The biggest one is close to 10000 buses.

benoit9126, 2024-01-23:
The number of buses in the SaaS application is limited on the server. It can't be modified by the user.

I am waiting on @alihamdan's PR #167 to release an alpha version of RLF as a non-SaaS application. With the good license key, you will be able to run load flow with an unlimited number of buses.

alihamdan, 2024-01-23:
> I am waiting on @alihamdan's PR #167 to release an alpha version of RLF as a non-SaaS application.

Working on one last error I got yesterday and we'll be ready to cut a pre-release.

benoit9126, 2024-01-24:
The version 0.7.0-alpha is available and allow the users to compute load flow locally without restriction on the bus count with the good license key. #171 

@YassineAbdelouadoud Please check your email to get your key. :-)

closed 2024-01-24 by benoit9126