RoseauTechnologies/Roseau_Load_Flow #214: BUG: Three-Phase Flexible production for nodes with 'abc' phases

YassineAbdelouadoud, 2024-04-11:
**Describe the bug**
When trying to access the `res_loads_flexible_powers` attribute of an `ElectricalNetwork` with a flexible `PowerLoad` connected to a node with `'abc'` phases (no neutral), the behaviour becomes unpredictable and any of the following can happen : 

- the attribute can be accessed but contains nans for the third phase : 
```
                                                         power
load_id                    phase                              
MVBus08988_production_flex ab    -882804.959531+     0.000000j
                           bc    -874293.640877+     0.000000j
                           ca               NaN+          NaNj
```

- a MemoryError: std::bad_alloc in self._cy_element.get_powers(self._n): 
```
  File "/home/yassine/miniconda3/envs/building-grid-model/lib/python3.11/site-packages/roseau/load_flow/network.py", line 1031, in res_loads_flexible_powers
    for power, phase in zip(load._res_flexible_powers_getter(warning=False), load.voltage_phases, strict=True):
                            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/yassine/miniconda3/envs/building-grid-model/lib/python3.11/site-packages/roseau/load_flow/models/loads/loads.py", line 351, in _res_flexible_powers_getter
    self._res_flexible_powers = self._cy_element.get_powers(self._n)
                                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "roseau/load_flow_engine/cy_engine.pyx", line 177, in roseau.load_flow_engine.cy_engine.CyDeltaFlexibleLoad.get_powers
MemoryError: std::bad_alloc
```

- a RuntimeError: cannot create std::vector larger than max_size() : 
```
  File "/home/yassine/miniconda3/envs/building-grid-model/lib/python3.11/site-packages/roseau/load_flow/network.py", line 1031, in res_loads_flexible_powers
    for power, phase in zip(load._res_flexible_powers_getter(warning=False), load.voltage_phases, strict=True):
                            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/yassine/miniconda3/envs/building-grid-model/lib/python3.11/site-packages/roseau/load_flow/models/loads/loads.py", line 351, in _res_flexible_powers_getter
    self._res_flexible_powers = self._cy_element.get_powers(self._n)
                                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "roseau/load_flow_engine/cy_engine.pyx", line 177, in roseau.load_flow_engine.cy_engine.CyDeltaFlexibleLoad.get_powers
RuntimeError: cannot create std::vector larger than max_size()
```

**To Reproduce**
Minimal Working Example to understand the problem

```python
from roseau.load_flow import *

en = ElectricalNetwork.from_catalogue(name="MVFeeder210", load_point_name="Winter")

power = 1.e6
u_max = 21000.
u_min = 19000.
u_up = u_min + (u_max - u_min) * 0.8

fp = FlexibleParameter.p_max_u_production(u_up=u_up,
                                          u_max=u_max,
                                          s_max=power,
                                          alpha_proj=50.,
                                          alpha_control=50.,)
bus = list(en.buses.values())[10]
prod_load = PowerLoad(id=bus.id + '_production_flex',
                      bus=bus,
                      powers=Q_([-power, -power, -power], 'VA'),
                      flexible_params=[fp, fp, fp])

en.solve_load_flow(max_iterations=10, tolerance=1.e-4)
flexible_powers = en.res_loads_flexible_powers
print(flexible_powers)
```

**Expected behavior**
Reliable access to the flexible load results including the 'ca' phase. 
It can be noted that this works reliably with 'abcn' nodes or when the PowerLoads are created independently between each phase.

**Version details**
This bug is present in both 0.8.0 and 0.7.0a0 versions.

```
System Information
------------------
python               3.11.7.final.0
os                   linux
os_name              posix
machine              x86_64
Installed Dependencies
----------------------
pandas               2.2.1
numpy                1.26.3
geopandas            0.14.2
shapely              2.0.2
regex                2023.12.25
pint                 0.23
platformdirs         4.2.0
certifi              2024.2.2
roseau-load-flow-engine 0.13.0
```

Saelyos, 2024-04-11:
Thanks for the report, there was indeed a bug with delta flexible loads. It will be fixed in the next version.

benoit9126, 2024-04-11:
@YassineAbdelouadoud It should be solved in the version 0.8.1: https://github.com/RoseauTechnologies/Roseau_Load_Flow/releases/tag/v0.8.1

closed 2024-04-11 by benoit9126